mbari-org/auvstatus #79: Change Old GPSFix limit to 3

beroe, 2023-10-05:
Currently at 4, pending incorporation of Carlos' fix for duplicates.

beroe, 2023-10-05:
Code is at  #L1945

carueda, 2023-10-05:
@beroe Not sure I follow, but, just in case, the new check for duplicates is in place in TethysDash since we had that discussion; it's just that the branch with such adjustments hasn't been merged into main. BTW, I may be doing that merge this week as all seems OK with GPSFix events, right?

beroe, 2023-10-05:
Hi Carlos. Thanks, I wasn't clear if "merged into main" was when it would be live for the vehicles. I just noticed a pair of fixes that were  close together, so I thought I needed to switch it back to 4 events instead of just 3...
`
[image]
`

beroe, 2023-10-05:
Thanks Carlos. Yes, there are two that are about a minute apart which was not useful for my dead-reckoning. I think I might have to change the approach, and look at the time difference from the most recent fix, before deciding which one to use. Normally it is not an issue, but when the vehicle sits on the surface for a while it throws off the average.

closed 2023-10-05 by beroe