mbari-org/auvstatus #118: Add a tiny dot on the sparkline to mark where it has surfaced

beroe, 2025-07-22:
Where to get that info all at once? Comms records? Maybe make a bar for when it's on the surface. Usually you can tell, but not when it is like the attached profile. 

[image]

carueda, 2025-07-23:
@beroe Good question. It seems the query (or queries) would involve a combination/aggregation of the direct and sat comms events that have occurred during a given timeframe, to then estimate the being-at-the-surface time windows.

```
[
  {
     start: ..,
     end: ..,
  }, 
  {
     start: ..,
     end: ..,
  }, 
  ..
}
```

beroe, 2025-07-23:
I guess we could also include GOS fixes and Argo, in case both comms methods are not working then we would still know it was on the surface..?

carueda, 2025-07-23:
Yes, GPS fixes and Argos also look like good indicators.

beroe, 2025-07-25:
Try using height 1 opaque lines separate levels not overlapping...

commit a126108e (2025-07-29): Version to reflect spark comms. Closes #118
closed 2025-07-29 by beroe